AislerHQ/aisler-support #8: Can't upload Altium 4-layer PCB correct

Henrik-Wo, 2021-02-23:
Project created with Altium 21.1.1
4-layer stack:
   1. signal layer
   2. power plain
   3. power plain
   4. signal layer

-> the use of "power plains" is not optional!

### Problem:

1. If I upload the *.PcbDoc, as suggested, your tool only recognise the top and bottom layer (-> 2-layer)
2. If I create a Gerber file and upload it, the internal planes are interpreted as positive even though they are negative. At least that is how they are displayed in the Board Inspector!

Libermentix, 2021-02-23:
Hi @Henrik-Wo 
thanks for reporting, do you mind sending us the corresponding aisler project link to felix@aisler.net so that we can have a closer look at your PcbDoc file?

PatrickFranken, 2021-02-23:
The issue is caused by the power planes used for the two internal layers. Please update the layer type from "power" to "signal" and both Gerber and Altium import will work fine.

closed 2021-02-23 by PatrickFranken

Henrik-Wo, 2021-02-23:
yes, that's what I thought, but I need power layers in the PCB design to be able to adjust my impedances correctly! Wouldn't it make sense if you could deal with negative layers or at least place a visible hint somewhere, so that I can already adjust to it during layout?