AislerHQ/aisler-support #10: Eagle >DATE_TIME format

UweBonnes, 2021-12-08:
Hello,

while local eagle uses the german time format, Aisler rendering uses the american time format. This uses substantial more area and my coworkers can be fooled by e.g. 12/8/2021 vs 08.12.2021 (august vs december)
See attached JPG.
Please think about a better solution!

[image]

UweBonnes, 2023-04-05:
Yes. Look at eg. https://aisler.net/p/MATULCMD

T-Krug, 2023-05-22:
This is unfortunately a no fix.  We have a server where the export happens automatically we cannot adjust it on a customer/location base.  

You can follow this Guide and create a Manufacturing Output that is also suitable for assembly locally thus your personal settings will be applied. https://community.aisler.net/t/exporting-a-custom-gerber-file-from-eagle/2309

closed 2023-05-22 by T-Krug