AislerHQ/aisler-support #13: Half laser name of design in stencils

UweBonnes, 2021-12-23:
It would be very helpfull if the name of the design would be visible on the stancil half-lasered!

PatrickFranken, 2022-01-27:
Good point! It is probably not possible to do it half lasered as this would be to time consuming, however we will add it as cut-through text. I added an internal issue #138 to follow this request.

closed 2022-01-27 by PatrickFranken

UweBonnes, 2022-01-27:
I do not know is "half lasered" is the same as "angelasert" in https://www.multi-circuit-boards.eu/leiterplatten-design-hilfe/smd-schablonen.html. I meant "angelasert"

PatrickFranken, 2022-02-01:
And exactly that "angelastert" is super time consuming why we would rather decide for cut -through as we do for the other text on the stencil. Anyway, I guess that is more of a minority. We will definitely add the project name.